corundum/corundum #236: Is it necessary to instantiate RAM in order to do DMA writes from the app block?

harris-chris, 2025-04-24:
Hello,

The interface for DMA reads/writes in the `mqnic_app_block.v` file has `dma_addr`, `ram_sel` and `ram_addr` signals, suggesting that we don't directly write to this interface (there is no `data`-like signal to directly write to), but rather use it to transfer information between an existing RAM instance and the DMA streams.

If so, in order to (say) write to DMA:
- instantiate a RAM block 
- write some data to this RAM block
- then use the DMA interface, where `ram_sel` and `ram_addr` are pointing to where we just wrote the data in our RAM block

This seems to be how things are done in eg `dma_bench.v` so I assume that's correct but I thought I'd check.

Thanks!

alexforencich, 2025-04-28:
Yeah, that's how it works right now.

I'm considering reworking this for the "next generation" version of Corundum (in SV, built on top of taxi).  Not sure yet what makes the most sense...I am considering moving back to using AXI and having effectively an AXI-PCIe bridge instead of a PCIe DMA engine.  This would certainly make things easier on Versal and Zynq and such.  But there might be other options as well.

harris-chris, 2025-04-29:
Thanks Alex!

closed 2025-04-29 by harris-chris