corundum/corundum #243: Alveo U280 network bandwidth

lifeis-hard, 2025-06-19:
Hi, thank you for this amazing project.

I have built and tested corundum default design (w/o dma_app_bench) on two Alveo U280 FPGAs, and performed several network tests.
However, I obtained results detailed as below, which is pretty much less than near 100Gbps, and I wonder if this amount of degarde in performance is expected or predictable.

[BW test]
- MTU 1.5KB
`iperf3` with single core: ~14Gbps
`iperf3` with 8 cores: ~30Gbps
`iperf3` with 16 cores: ~40Gbps
`qperf tcp_bw`: ~1.5GB/s
- MTU 9KB
`iperf3` with 8 cores: ~50Gbps
`iperf3` with 16 cores: ~60Gbps

I have performed `iperf3` in two ways to correctly execute with multiple cores as below:
- `iperf3 -P 8`: I used `iperf3` version 3.16, therefore I expect it to properly support mutli threading.
- `taskset -c $i iperf3 -s -A $i`: Set CPU affinity as `iperf3` suggests. `$i` in `{0..7}`

[Setup]
- Vivado 2022.2
- The tested servers are equipped with 2 U280 cards, both cards built with default corundum design.
- One server has 8 cores, while the other has 16 cores.
- The two FPGAs are connected directly via 100G cable.

Seems the achievable network BW is about ~40Gbps with MTU 1.5KB, in my current testbed.
Would this amount of performance is in the boundary corundum offers?
Or will there be any missing/need-to-check performance-related features I should try with?

Any type of support would be greatly appreciated.
Thank you.

alexforencich, 2025-06-19:
What's the model of the server and the CPU?

alexforencich, 2025-06-19:
And also what version of linux are you running?

lifeis-hard, 2025-06-19:
Two servers are both custom servers, and the CPU models are as below:
- 8 core server: i7-9700K, Linux kernel 5.4.0-198-generic
- 16 core server: i7-10700K, Linux kernel 5.4.0-77-generic

I also attached `lspci -vvd 1234:1001` output as below, in case it may be helpful:

```
01:00.0 Ethernet controller: Device 1234:1001
        Subsystem: Xilinx Corporation Device 9118
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Region 0: Memory at 50000000 (64-bit, prefetchable) [size=16M]                                                                                                                                                                               Capabilities: [40] Power Management version 3                                                                                                                                                                                                        Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [60] MSI-X: Enable+ Count=64 Masked-
                Vector table: BAR=0 offset=00010000                                                                                                                                                                                                          PBA: BAR=0 offset=00018000                                                                                                                                                                                                           Capabilities: [70] Express (v2) Endpoint, MSI 00                                                                                                                                                                                                     DevCap: MaxPayload 1024 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us                                                                                                                                                                               ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 75.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-                                                                                                                                                                                                    RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 8GT/s, Width x16, ASPM not supported
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 8GT/s (ok), Width x16 (ok)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range BC, TimeoutDis+, NROPrPrP-, LTR-
                         10BitTagComp-, 10BitTagReq-, OBFF Not Supported, ExtFmt-, EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-, TPHComp-, ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete+, EqualizationPhase1+
                         EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout+ AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [1c0 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn-, PerformEqu-
                LaneErrStat: LaneErr at lane: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15
        Capabilities: [1f0 v1] Virtual Channel
                Caps:   LPEVC=0 RefClk=100ns PATEntryBits=1
                Arb:    Fixed- WRR32- WRR64- WRR128-
                Ctrl:   ArbSelect=Fixed
                Status: InProgress-
                Port Arbitration Table [500] <?>
                VC0:    Caps:   PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
                        Arb:    Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
                        Ctrl:   Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
                        Status: NegoPending- InProgress-
        Kernel driver in use: mqnic
```

The `lspci` output seems to be identical in both servers except the line `Region 0: Memory at ${address} ~ `.

lifeis-hard, 2025-06-19:
Thanks.
I found that the 8 core server's CPU was i7-9700 not i7-9700K, which makes it even worse.

But would there be nothing else to check or try rather than changing the host CPU part?
I just want to make sure that I have tried most of the available options before buying a new CPU.